OpenCyphal/pycyphal #43: Module fails to import under Windows

CaptainJK, 2018-09-30:
Hello.
This is probably a minor issue. In Windows, uavcan being imported from within drive, other than the one, containing python lib, throws exception:

```
c:\users\captain\appdata\local\programs\python\python36-32\lib\site-packages\uavcan\dsdl\common.py in pretty_filename(filename)
     39     '''Returns a nice human readable path to 'filename'.'''
     40     a = os.path.abspath(filename)
---> 41     r = os.path.relpath(filename)
     42     return a if '..' in r else r
     43

c:\users\captain\appdata\local\programs\python\python36-32\lib\ntpath.py in relpath(path, start)
    584         if normcase(start_drive) != normcase(path_drive):
    585             raise ValueError("path is on mount %r, start on mount %r" % (
--> 586                 path_drive, start_drive))
    587
    588         start_list = [x for x in start_rest.split(sep) if x]

ValueError: path is on mount 'c:', start on mount 'D:'
```

Quite eloquent, I guess. The fix might be to catch ValueError and return filename as is.

pavel-kirienko, 2018-09-30:
Yes indeed. Care to submit a PR?

pavel-kirienko, 2018-09-30:
Fixed in https://github.com/UAVCAN/pyuavcan/pull/44

closed 2018-09-30 by pavel-kirienko