OpenCyphal/pycyphal #45: Exception in callbacks are caught by PyUAVCAN

antoinealb, 2018-10-01:
Sometimes I hit a bug or an exception in a callback. However this never hits my debugger because UAVCAN  catches them and ignores them.

Example: https://github.com/UAVCAN/pyuavcan/blob/master/uavcan/node.py#L217-L220

I would suggest passing a flag to the node or the spin method, something like `node.spin(catch_callback_exceptions=False)` to let advanced users use exceptions like they want.

What do you think? Maybe I am missing something on this design choice?

pavel-kirienko, 2018-10-04:
Pyuavcan does that to ensure that exceptions that occur in a handler do not prevent the library from invoking other handlers. Adding a flag as you suggested seems sensible. Perhaps it's time for a pull request?

antoinealb, 2018-10-04:
PR opened: #46

antoinealb, 2018-10-05:
Closed in #46

closed 2018-10-05 by antoinealb