OpenCyphal/pycyphal #55: Idea: merge several repositories into single repository

windelbouwman, 2019-04-08:
This is idea for an improvement, please feel free to close this ticket if required.

Currently there are several interlinked git repositories which depend upon eachother by the use of git submodules. It often makes sense to split a codebase into several git repositories, but it also has its downsides. Some difficulties that might occur:
- a change is spread over mutliple repositories
- Care must be taken to keep different repositories compatible.
- When checking out gui_tool with full submodules, I have multiple copies of pyuavcan occuring on my system making it potentially confusing which directory is used.

In order to reduce the amount of repositories, I would suggest to regroup several repositories into a single repository like this:
- `uavcan`: a merger of `pyuavcan`, `specification`, `libuavcan`, 'dsdl`, `libuavcan`, `libcanard`. My suggestion is to bundle specification, compiler, standard definition and reference implementation, since these are tightly coupled and must be in sync at all times.
- `gui_tool` and `uavcan.rs`: these are more building upon the `uavcan` repository. Optionally those could be subfolders of the bigger repository.

Let me know what you think!

pavel-kirienko, 2019-04-08:
Thanks.

> a change is spread over mutliple repositories
> Care must be taken to keep different repositories compatible.

These are non-issues, because there is no requirement to keep things compatible between themselves. The only requirement is to keep them compliant with the specification.

> When checking out gui_tool with full submodules, I have multiple copies of pyuavcan occuring on my system making it potentially confusing which directory is used.

We sort of fixed this by separating the DSDL front-end into a separate project: https://github.com/UAVCAN/pydsdl

Merging things the way you described would be a mistake, since they are intended to be well-isolated from each other. For example, there is no way pyuavcan can possibly depend on anything defined in libcanard.

closed 2019-04-08 by pavel-kirienko

windelbouwman, 2019-04-08:
> Merging things the way you described would be a mistake, since they are intended to be well-isolated from each other. For example, there is no way pyuavcan can possibly depend on anything defined in libcanard.

Merging different pieces of code into a single repository does not mean they are not isolated. Putting the different pieces into seperate directories is also a form of isolation. While you are right that pyuavcan does not depend upon libcanard (and vice versa), it would be handy to have some demo setups with the combination of the two, for example. This could then be automated and checked in travis, to check if the two implementation are compatible.

pavel-kirienko, 2019-04-08:
True, but keeping unrelated codebases in the same repository makes limited sense.

windelbouwman, 2019-04-08:
That is indeed the point, a couple of uavcan repositories are related and linked by submodules. I would prefer subfolders over submodules to keep things simpler.

kjetilkjeka, 2019-04-13:
The diplomatic thing to say in regards to this is that it's a question of preference: https://redfin.engineering/well-never-know-whether-monorepos-are-better-2c08ab9324c0

I think both things could work for the Uavcan organization, but I'm also very happy with what we're currently doing. As there are not many objective reasons for preferring monorepos, and the core contributors seem to be happy, I don't think it will be wise to change anything.

windelbouwman, 2019-04-13:
Nice post, I agree that keeping the core devs happy is the most important thing here. Just want to pitch the idea, since it was new to me as well, some years back.