OpenCyphal/pycyphal #62: SLCAN driver based on PySerial

pavel-kirienko, 2019-06-14:
Run the RX logic in a separate thread. Raise an error during instantiation if PySerial is not available. Support the SLCAN command line extensions as documented in the [Babel datasheet](https://files.zubax.com/products/com.zubax.babel/Zubax_Babel_Datasheet.pdf).

This is crucial for supporting non-Linux-based OS (on Linux there's SocketCAN which covers SLCAN as well). Many low-cost USB-CAN adapters out there support SLCAN.

pavel-kirienko, 2019-09-16:
Closing because this functionality is a subset of #63.

silverv, 2022-09-19:
Python-CAN is occasionally broken so we have to implement this.

silverv, 2022-09-19:
Old implementation: https://github.com/dronecan/pydronecan/blob/master/dronecan/driver/slcan.py

SLCAN protocol specification: https://files.zubax.com/products/com.zubax.babel/Zubax_Babel_Datasheet.pdf