OpenCyphal/pycyphal #63: Python-CAN media driver

pavel-kirienko, 2019-06-14:
Basically, we need this reimplemented for PyUAVCAN v1.0: https://github.com/UAVCAN/pyuavcan/pull/53

@windelbouwman @adolfogc could use your help here!

adolfogc, 2019-06-18:
Hi @pavel-kirienko, I'm in the process of studying the UAVCAN v1.0 specification and getting familiar with PyUAVCAN's v1.0 branch. After this, I would like to start working on adding support for SocketCAN using Python-CAN.

pavel-kirienko, 2019-06-19:
@adolfogc this is cool. Please observe that there is a SocketCAN media implementation in the upstream already; it makes sense to focus on other devices supported by Python-CAN.

pavel-kirienko, 2019-09-16:
Hey @adolfogc is there still interest?

jxltom, 2020-04-22:
Since python-can supports socketcan natively, it might make sense that the current socketcan media driver can be removed.

pavel-kirienko, 2021-01-18:
Implemented in #123

closed 2021-01-18 by pavel-kirienko