OpenCyphal/pycyphal #64: Test against GNU/Linux and Windows using containers

pavel-kirienko, 2019-06-27:
https://forum.uavcan.org/t/pyuavcan-design-thread/504/7?u=pavel.kirienko

pavel-kirienko, 2019-07-19:
Scott said:

> Just **use a docker container**. We have a docker hub organization setup already and Nunavut and libuavcan already build in containers.

But how does one load a kernel module in a containerized environment?

I just tried AppVeyor as was suggested in the linked thread, turns out they don't support vcan either:

```
modprobe: FATAL: Module vcan not found in directory /lib/modules/4.15.0-1027-gcp
```

We need to do something about this. I would really like to avoid any options involving self-hosted build servers.

thirtytwobits, 2019-07-26:
Yeah, vcan will have to be natively available. You could use buildkite+EC2 or even AWS codebuild + EC2. We might have to build a new AMI but that would be worth the effort given that I could reuse the AMI for libuavcan builds and reduce my reliance on PI workers.

pavel-kirienko, 2019-07-26:
Nvm, RTFM helped. Turns out they ship `vcan.ko` in extra packages. This commit made it work both on AppVeyor and Travis: https://github.com/UAVCAN/pyuavcan/commit/15e03415de88150723ad055c525723928667ebf1

Windows is up next. Would anybody like to volunteer?

pavel-kirienko, 2019-08-27:
Fixed in #71

PR #71 (2019-08-27): Windows support
Works on Windows 10. The CI rigging has been carefully updated with Windows-specific hacks (sigh).

For some reason, I couldn't make PIP install my CLI shims properly, so the command-line tool has to be invoked via `python -m pyuavcan` instead of `pyuavcan` or `uc`. We'll need someone well-versed in Microsoft ways to look into that later, I guess.

Closes #64.
closed 2019-08-27 by pavel-kirienko