OpenCyphal/pycyphal #95: Tests fail with PyTest v5.x

pavel-kirienko, 2020-02-19:
I tried upgrading to PyTest v5.3, but Nunavut stopped working when invoked by the testing suite:

```
NotImplementedError("Can't perform this operation for unregistered loader type")
Traceback (most recent call last):

  <...irrelevant...>

  File "/home/travis/build/UAVCAN/pyuavcan/pyuavcan/dsdl/_compiler.py", line 197, in generate_package
    namespace_output_stem='__init__')

  File "/home/travis/virtualenv/python3.7-dev/lib/python3.7/site-packages/nunavut/lang/__init__.py", line 225, in __init__
    self._config = self._load_config(*additional_config_files)

  File "/home/travis/virtualenv/python3.7-dev/lib/python3.7/site-packages/nunavut/lang/__init__.py", line 208, in _load_config
    resources = [r for r in pkg_resources.resource_listdir(__name__, '.') if r.endswith('.ini')]

  File "/home/travis/virtualenv/python3.7-dev/lib/python3.7/site-packages/pkg_resources/__init__.py", line 1164, in resource_listdir
    resource_name

  File "/home/travis/virtualenv/python3.7-dev/lib/python3.7/site-packages/pkg_resources/__init__.py", line 1442, in resource_listdir
    return self._listdir(self._fn(self.module_path, resource_name))

  File "/home/travis/virtualenv/python3.7-dev/lib/python3.7/site-packages/pkg_resources/__init__.py", line 1484, in _listdir
    "Can't perform this operation for unregistered loader type"

NotImplementedError: Can't perform this operation for unregistered loader type
```

It's super weird that the exception comes from `pkg_resources`. It does not seem to be involved here at all.

Nunavut works fine outside of PyTest. I googled around a bit but could not find a quick solution, so I decided to downgrade back to PyTest v4.6 for now.

thirtytwobits, 2020-02-19:
Ack. Yeah, I use pkg_resources to load templates from the package. I should probably switch to importlib.

jxltom, 2020-05-26:
Sorry update the I forgot to remove the closing keyword in the PR. This issue is not solved actually, please reopen  this. Sorry for the inconvenience.

pavel-kirienko, 2020-05-26:
Indeed, I didn't notice that it got closed automatically. Thanks for raising this.

pavel-kirienko, 2020-10-17:
@thirtytwobits should we transfer this issue over to Nunavut?

pavel-kirienko, 2020-10-17:
No. Nunavut v0.3.9 with PyTest 6.1 just work. This is my preferred kind of problem that just solves itself.

pavel-kirienko, 2020-10-17:
Nvm it is still there.