OpenCyphal/pycyphal #97: Branch legacy-v0 actually contains code for v1

jxltom, 2020-02-20:
It looks like the ```master``` branch is v0 version and ```legacy-v0``` branch is v1 version, which is a little bit confusing.

pavel-kirienko, 2020-02-20:
This is not the case: master is v1, and legacy-v0 is v0. Could you please tell us what made you believe otherwise so that we could update the docs?

jxltom, 2020-02-21:
Because it looks like https://github.com/UAVCAN/pyuavcan/blob/master/pyuavcan/VERSION is tagged as ```0.6.2``` while https://github.com/UAVCAN/pyuavcan/blob/legacy-v0/uavcan/version.py is tagged as ```1.0.0.dev32```, which is same as https://github.com/UAVCAN/pyuavcan/blob/1.0.0.dev32/uavcan/version.py.

That's the thing confusing me.

pavel-kirienko, 2020-02-21:
Ah, yes, confusing indeed. Years ago we made a mistake of releasing a v1.0devX version too early, and now it's too late to change it, so we are stuck in this now. You should understand though that the version of an IMPLEMENTATION has nothing to do with the version of the SPECIFICATION that it implements. Like, for example, the latest version of libusb is v1.0.23, but it does not imply that it is incompatible with USB v3.1, they are just versioned independently. Same here: PyUAVCAN v0.6 supports UAVCAN v1.x.

I added a few notes into the v0's README, hope that should make things clearer now.

closed 2020-02-21 by pavel-kirienko

jxltom, 2020-02-21:
Oh I get it now. So pyuavcan v0.6.2 supports uavcan protocol v1 while pyuavcan (uavcan actually) v1 supports uavcan protocol v0.

Thanks for the explanation. :+1:

pavel-kirienko, 2020-02-21:
[image]

pavel-kirienko, 2020-04-02:
#109 makes things clearer.

jxltom, 2020-04-03:
Nice