OpenCyphal/pycyphal #98: Can not generate dsdl packages for legacy public data types

jxltom, 2020-02-20:
I would like to generate dsdl packages for legacy v0 data types in https://github.com/UAVCAN/public_regulated_data_types/tree/legacy-v0. I installed ```pyuavcan``` 0.6.2 and the command ```pyuavcan dsdl-generate-packages public_regulated_data_types/uavcan``` returns ```Error: FileNameFormatError: ....../public_regulated_data_types/uavcan/CoarseOrientation.uavcan: Invalid file name```

pavel-kirienko, 2020-02-20:
It's not possible [by design](https://forum.uavcan.org/t/stockholm-summit-recap/170). The syntax is slightly different, but that is not the major culprit -- that would be the fact that v0 DSDL has a tail array optimization (TAO) and v1 doesn't. This difference breaks binary compatibility.

If you want to re-use types from v0, you could just manually update them to make the v1 parser happy -- just add version numbers. For example, the source file `uavcan/CoarseOrientation.uavcan` would become `uavcan/CoarseOrientation.1.0.uavcan`. A reference to a composite type like `uavcan.CoarseOrientation foo` would become `uavcan.CoarseOrientation.1.0 foo`.

Why do you need v0 types anyway?

jxltom, 2020-02-21:
Because uavcan v1 is not stable yet, some libraries such as libuavcan v1 still lacks core features such as dsdl generator.

So for now I'm going to use v0 instead until v1 is stable. I think libcanard v0 or libuavan v0 requires v0 types as well, right? For example, the master branch of libuavcan in https://github.com/UAVCAN/libuavcan is using v0 types as linked as ```dsdl@192295c```. That's the reason why I need v0 types.

Let's me know if I'm wrong, thanks!

pavel-kirienko, 2020-02-21:
You are correct -- v0 implementations require v0 types.

closed 2020-02-21 by pavel-kirienko