OpenCyphal/pycyphal #108: Pypi versioning is unclear

thirtytwobits, 2020-04-02:
People are doing pip install and expecting a version compatible with what they've been using (v0) but are instead getting the v1 versions. We need to fix this in one of two ways:

1. delete the v1 versions released as >= 0.2 and start releasing these as >= 1.x
2. Create a new pypi project that is pyuavcan1 (or something similar).

pavel-kirienko, 2020-04-02:
v0 is `uavcan` on PyPI
v1 is `pyuavcan` on PyPI

thirtytwobits, 2020-04-02:
so v0 versions are 1.x and v1 versions are 0.x? This is really confusing.

pavel-kirienko, 2020-04-02:
Yes, unfortunately https://github.com/UAVCAN/pyuavcan/issues/97. We could any of the following, as I see it:

- Unrelease all v1.x.x, and re-release the latest v1.0.0.dev32 as v0.0.1.
- Keep things as-is.
- Release PyUAVCAN v1.0 but with a note that API stability is not yet guaranteed until v1.1.

thirtytwobits, 2020-04-02:
what about changing pyuavcan to use v1.1.x.unstable?

PR #109 (2020-04-02): Fix the confusing versioning problem per #108
Fixes #108.

Would anybody like to volunteer to help us with #76?
closed 2020-04-02 by pavel-kirienko