OpenCyphal/pycyphal #113: UDP localhost transport is not working on macOS

pavel-kirienko, 2020-05-23:
@arrowcircle says that the UDP transport is not working on localhost on macOS in https://github.com/UAVCAN/Yukon/pull/46#issuecomment-628644462.

I am unable to reproduce because I don't have any macOS computers within reach. Maybe it's worth looking into veth tunnels or similar alternatives if there are any on macOS, like https://stackoverflow.com/questions/58080648/broadcast-packets-do-not-propagate-through-a-linux-veth-tunnel/58118503.

If a solution is found, it should be described in the docs. If not, it should also be stated there as a known issue.

arrowcircle, 2020-05-25:
Do we have any test suite or example, that fails? Or should we extract it from Yukon?

pavel-kirienko, 2020-05-25:
Maybe you could try the demo https://pyuavcan.readthedocs.io/en/latest/pages/usage.html

closed 2020-12-22 by pavel-kirienko