OpenCyphal/pycyphal #122: Nunavut-related regression

pavel-kirienko, 2020-08-06:
With Nunavut 0.3.x we get:

```
2020-08-06 12:27:53 87443 INFO     pyuavcan._cli._main: Unhandled exception: test saturated was already defined.
Traceback (most recent call last):
  File "/home/pavel/.local/lib/python3.8/site-packages/pyuavcan/_cli/_main.py", line 24, in main
    exit(_main_impl())
  File "/home/pavel/.local/lib/python3.8/site-packages/pyuavcan/_cli/_main.py", line 50, in _main_impl
    result = args.func(args)
  File "/home/pavel/.local/lib/python3.8/site-packages/pyuavcan/_cli/_main.py", line 149, in execute
    return cmd.execute(args, subsystems)
  File "/home/pavel/.local/lib/python3.8/site-packages/pyuavcan/_cli/commands/dsdl_generate_packages.py", line 118, in execute
    gpi_list = self._generate_dsdl_packages(source_root_namespace_dirs=inputs,
  File "/home/pavel/.local/lib/python3.8/site-packages/pyuavcan/_cli/commands/dsdl_generate_packages.py", line 186, in _generate_dsdl_packages
    gpi = pyuavcan.dsdl.generate_package(root_namespace_directory=ns,
  File "/home/pavel/.local/lib/python3.8/site-packages/pyuavcan/dsdl/_compiler.py", line 202, in generate_package
    generator = nunavut.jinja.Generator(namespace=root_ns,
  File "/home/pavel/.local/lib/python3.8/site-packages/nunavut/jinja/__init__.py", line 527, in __init__
    self._add_filters_and_tests(additional_filters, additional_tests)
  File "/home/pavel/.local/lib/python3.8/site-packages/nunavut/jinja/__init__.py", line 638, in _add_filters_and_tests
    raise RuntimeError('test {} was already defined.'.format(name))
RuntimeError: test saturated was already defined.
```

thirtytwobits, 2020-08-06:
looking...

thirtytwobits, 2020-08-06:
The short answer is you can simply remove the `saturated` filter from pyuavcan since this is now included in Nunavut. The in Nunavut is to simply log if built-in filters conflict with additional_[filters|tests|globals] and take the user-supplied version instead. Let me know if this is urgent to fix in Nunavut or if you want to fix it in pyuavcan. I've got the two other features in-flight to support the C-serialization PR so it would be a week or two before I got to this.

PR #128 (2020-11-01): Conform to Specification v1.0-beta
We are not quite release-ready yet. This changeset makes the library compliant and tested against v1.0-beta, but there are a couple of other things that need to be addressed before we can make the first release.

Fixes #125 by updating to a newer version of PyDSDL (https://github.com/UAVCAN/pydsdl/issues/41, https://github.com/UAVCAN/pydsdl/issues/46).

Fixes #122
closed 2020-11-01 by pavel-kirienko