OpenCyphal/pycyphal #129: after pip install there is a missed dependency from cobs

pat1, 2020-11-02:
```
Error: RuntimeError: Subsystem factory 'TransportFactory' for command 'subscribe' has failed: No module named 'cobs'
2020-11-02 18:15:31 16892 INFO     pyuavcan._cli._main: Unhandled exception: Subsystem factory 'TransportFactory' for command 'subscribe' has failed: No module named 'cobs'
Traceback (most recent call last):
  File "/home/pat1/compilazioni/pyuavcan-v1/lib/python3.7/site-packages/pyuavcan/_cli/_main.py", line 139, in execute
    ss = sf.construct_subsystem(args)
  File "/home/pat1/compilazioni/pyuavcan-v1/lib/python3.7/site-packages/pyuavcan/_cli/commands/_subsystems/transport.py", line 73, in construct_subsystem
    context = _make_evaluation_context()
  File "/home/pat1/compilazioni/pyuavcan-v1/lib/python3.7/site-packages/pyuavcan/_cli/commands/_subsystems/transport.py", line 109, in _make_evaluation_context
    pyuavcan.util.import_submodules(pyuavcan.transport)
  File "/home/pat1/compilazioni/pyuavcan-v1/lib/python3.7/site-packages/pyuavcan/util/_introspect.py", line 59, in import_submodules
    importlib.import_module(module_name)
  File "/home/pat1/compilazioni/pyuavcan-v1/lib64/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/pat1/compilazioni/pyuavcan-v1/lib/python3.7/site-packages/pyuavcan/transport/serial/__init__.py", line 207, in <module>
    from ._serial import SerialTransport as SerialTransport
  File "/home/pat1/compilazioni/pyuavcan-v1/lib/python3.7/site-packages/pyuavcan/transport/serial/_serial.py", line 18, in <module>
    from ._frame import SerialFrame
  File "/home/pat1/compilazioni/pyuavcan-v1/lib/python3.7/site-packages/pyuavcan/transport/serial/_frame.py", line 12, in <module>
    from cobs import cobs  # type: ignore
ModuleNotFoundError: No module named 'cobs'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pat1/compilazioni/pyuavcan-v1/lib/python3.7/site-packages/pyuavcan/_cli/_main.py", line 24, in main
    exit(_main_impl())
  File "/home/pat1/compilazioni/pyuavcan-v1/lib/python3.7/site-packages/pyuavcan/_cli/_main.py", line 50, in _main_impl
    result = args.func(args)
  File "/home/pat1/compilazioni/pyuavcan-v1/lib/python3.7/site-packages/pyuavcan/_cli/_main.py", line 141, in execute
    raise RuntimeError(f'Subsystem factory {type(sf).__name__!r} for command {cmd.names[0]!r} '
RuntimeError: Subsystem factory 'TransportFactory' for command 'subscribe' has failed: No module named 'cobs'
```

after 
```
pip3 install cobs
```
all is right

pavel-kirienko, 2020-11-02:
Thank you. This issue duplicates https://github.com/UAVCAN/pyuavcan/issues/114

closed 2020-11-02 by pavel-kirienko

aentinger, 2020-11-13:
I don't think this issue duplicates #114 as I experiences the same error and fix even with prior execution of `pip install pyuavcan[transport_can_pythoncan,transport_serial,cli]
`.

pavel-kirienko, 2020-11-13:
Can't reproduce locally. Do you happen to have an exact sequence of actions that yields this result?

aentinger, 2020-11-16:
The issue stems probably from me not having installed `cobs` on my system so far, maybe it needs to be added to the tool dependencies?

pavel-kirienko, 2020-11-16:
It is a dependency ofc. It is automatically pulled when you select the `transport_serial` package extra.

aentinger, 2020-11-17:
I darkly remember that this was a trap I already fell. Can you not just pull it in automatically whenever one instally `pyuavcan`, given that it's apparently needed anyway?

pavel-kirienko, 2020-11-18:
That is not a sensible option because the core library does not really need COBS. Only the serial transport needs it, hence it's a transport-specific dependency.

aentinger, 2020-11-18:
I disgress. How can it be that get this error even if I'm not using serial transport at all?

pavel-kirienko, 2020-11-18:
Are you sure you are still getting this error with v1.1.dev3? It's supposed to be fixed in #132.

aentinger, 2020-11-18:
I :bow: and apologise. I've cleaned my system and reinstalled everything from scratch, `cobs` is no longer needed :+1: