OpenCyphal/pycyphal #130: DSDL compiler fails if the DSDL namespace is empty

pavel-kirienko, 2020-11-02:
The tuple unpacking here fails if the list of data types is empty:

https://github.com/UAVCAN/pyuavcan/blob/ac2ed338d215e62d125d6858cd44bb3154336c31/pyuavcan/dsdl/_compiler.py#L188

https://forum.uavcan.org/t/unknown-directive-sealed-error-when-trying-to-generate-from-dsdl/955/16?u=pavel.kirienko

aentinger, 2020-11-11:
Same issue here, I'm not sure how to proceed, I've cloned `public_regulated_data_types` locally and call pyuavcan like this `pyuavcan -v dsdl-gen-pkg public_regulated_data_types/ --output test/`but it's still not working.

```bash
pydsdl._data_type_builder.UndefinedDataTypeError: /home/alex/projects/107-systems/public_regulated_data_types/reg/drone/phy/acoustics/Note.0.1.uavcan:6: Data type uavcan.si.unit.frequency.Scalar.1.0 could not be found in the following root namespaces: {'public_regulated_data_types'}.  Please make sure that you specified the directories correctly.
```

Sorry to bother you about this @pavel-kirienko but what parameter sequence am I supposed to feed to pyuavcan?

pavel-kirienko, 2020-11-11:
You were close:

```diff
- pyuavcan -v dsdl-gen-pkg public_regulated_data_types/ --output test/
+ pyuavcan -v dsdl-gen-pkg public_regulated_data_types/uavcan --output test/
```

commit cf10193f (2020-11-13): Fix #130
PR #132 (2020-11-14): Fixes for multiple recently reported minor issues
There are no major changes in this pull request. It mostly consists of bug fixes and minor UX improvements suggested over the last months on the forum. Also, it finally brings the continuous deployment feature which should make releases much more regular and much less tiresome. Each published release is also automatically tagged in the upstream for change control.

Fix #114
Fix #117
Fix #130
Fix #120
Fix #118
Fix #116
Fix #115
Fix #76 :snake: 

Read the docs for this changeset at https://pyuavcan--132.org.readthedocs.build/en/132/

FYI @aenting [... cut]
commit f43afb54 (2020-11-14): Fixes for multiple recently reported minor issues (#132)

* Remove poorly supported formatting from README

* Add intersphinx integration for PyDSDL

* Working on release automation

* Fixing CD config

* CD config appears to be functional, restore the real test invocation

* Hide appveyor.yml

* Remove obsolete logic from codegen templates (no semantic changes)

* Fix #130

* Fix #115, fix #118

* Fix #120

* Fix #116

* Fix #114, fix #117: skip missing dependencies when constructing transports, support transport configuration via environment variable

* Fix test rigging and bump the version  [... cut]
closed 2020-11-14 by pavel-kirienko