OpenCyphal/pycyphal #131: Initial DSDL generation fails

TomDeRybel, 2020-11-03:
I'm trying to install and use pyuavcan-cli. It fails to generate the DSDL. These are the steps I took:

pip install pyuavcan[transport_can_pythoncan,transport_serial,cli]
--> OK

pyuavcan --version
--> pyuavcan 1.1.0.dev2

mkdir ~/uavcan (OR anything else! The name of the folder has no impact on the failure mode.)
cd ~/uavcan

pyuavcan -v dsdl-gen-pkg ~/uavcan --lookup https://github.com/UAVCAN/public_regulated_data_types/archive/master.zip          
2020-11-03 11:21:13  6850 INFO     pyuavcan._cli.commands.dsdl_generate_packages: Input DSDL root namespace directories: ['/home/tomr/uavcan']
2020-11-03 11:21:13  6850 INFO     pyuavcan._cli.commands.dsdl_generate_packages: Downloading the archive from 'https://github.com/UAVCAN/public_regulated_data_types/archive/master.zip' into '/tmp/pyuavcan-cli-dsdllx88qgdk/dsdl.zip'...
2020-11-03 11:21:14  6850 INFO     pyuavcan._cli.commands.dsdl_generate_packages: Extracting the archive into '/tmp/pyuavcan-cli-dsdllx88qgdk'...
2020-11-03 11:21:14  6850 INFO     pyuavcan._cli.commands.dsdl_generate_packages: Resource 'https://github.com/UAVCAN/public_regulated_data_types/archive/master.zip' contains the following root namespace directories: ['/tmp/pyuavcan-cli-dsdllx88qgdk/public_regulated_data_types-master/uavcan', '/tmp/pyuavcan-cli-dsdllx88qgdk/public_regulated_data_types-master/.vscode', '/tmp/pyuavcan-cli-dsdllx88qgdk/public_regulated_data_types-master/reg']
2020-11-03 11:21:14  6850 INFO     pyuavcan._cli.commands.dsdl_generate_packages: Lookup DSDL root namespace directories: ['/tmp/pyuavcan-cli-dsdllx88qgdk/public_regulated_data_types-master/uavcan', '/tmp/pyuavcan-cli-dsdllx88qgdk/public_regulated_data_types-master/.vscode', '/tmp/pyuavcan-cli-dsdllx88qgdk/public_regulated_data_types-master/reg']
2020-11-03 11:21:14  6850 INFO     pyuavcan._cli.commands.dsdl_generate_packages: Generating DSDL package PosixPath('/home/tomr/uavcan_namespace/uavcan') from root namespace PosixPath('/home/tomr/uavcan') with lookup dirs: ['/tmp/pyuavcan-cli-dsdllx88qgdk/public_regulated_data_types-master/.vscode', '/home/tomr/uavcan', '/tmp/pyuavcan-cli-dsdllx88qgdk/public_regulated_data_types-master/uavcan', '/tmp/pyuavcan-cli-dsdllx88qgdk/public_regulated_data_types-master/reg']
Error: RootNamespaceNameCollisionError: /home/tomr/uavcan: The name of this namespace conflicts with '/tmp/pyuavcan-cli-dsdllx88qgdk/public_regulated_data_types-master/uavcan'
2020-11-03 11:21:14  6850 INFO     pyuavcan._cli._main: Unhandled exception: /home/tomr/uavcan: The name of this namespace conflicts with '/tmp/pyuavcan-cli-dsdllx88qgdk/public_regulated_data_types-master/uavcan'
Traceback (most recent call last):
  File "/home/tomr/anaconda3/lib/python3.8/site-packages/pyuavcan/_cli/_main.py", line 24, in main
    exit(_main_impl())
  File "/home/tomr/anaconda3/lib/python3.8/site-packages/pyuavcan/_cli/_main.py", line 50, in _main_impl
    result = args.func(args)
  File "/home/tomr/anaconda3/lib/python3.8/site-packages/pyuavcan/_cli/_main.py", line 146, in execute
    return cmd.execute(args, subsystems)
  File "/home/tomr/anaconda3/lib/python3.8/site-packages/pyuavcan/_cli/commands/dsdl_generate_packages.py", line 111, in execute
    gpi_list = self._generate_dsdl_packages(source_root_namespace_dirs=inputs,
  File "/home/tomr/anaconda3/lib/python3.8/site-packages/pyuavcan/_cli/commands/dsdl_generate_packages.py", line 179, in _generate_dsdl_packages
    gpi = pyuavcan.dsdl.generate_package(root_namespace_directory=ns,
  File "/home/tomr/anaconda3/lib/python3.8/site-packages/pyuavcan/dsdl/_compiler.py", line 185, in generate_package
    composite_types = pydsdl.read_namespace(root_namespace_directory=str(root_namespace_directory),
  File "/home/tomr/anaconda3/lib/python3.8/site-packages/pydsdl/_namespace.py", line 154, in read_namespace
    _ensure_no_namespace_name_collisions(lookup_directories_sorted_path_list)
  File "/home/tomr/anaconda3/lib/python3.8/site-packages/pydsdl/_namespace.py", line 389, in _ensure_no_namespace_name_collisions
    raise RootNamespaceNameCollisionError('The name of this namespace conflicts with %r' % b, path=a)
pydsdl._namespace.RootNamespaceNameCollisionError: /home/tomr/uavcan: The name of this namespace conflicts with '/tmp/pyuavcan-cli-dsdllx88qgdk/public_regulated_data_types-master/uavcan'

pavel-kirienko, 2020-11-03:
This is expected and documented behavior. You have conflicting namespaces: `uavcan` and `uavcan`.

TomDeRybel, 2020-11-03:
Still having no luck.
mkdir ~/foobar
cd foobar
pyuavcan -v dsdl-gen-pkg ~/foobar --lookup https://github.com/UAVCAN/public_regulated_data_types/archive/master.zip

Still blows-up:
2020-11-03 15:30:47 12985 INFO     pyuavcan._cli.commands.dsdl_generate_packages: Input DSDL root namespace directories: ['/home/tomr/foobar']
2020-11-03 15:30:47 12985 INFO     pyuavcan._cli.commands.dsdl_generate_packages: Downloading the archive from 'https://github.com/UAVCAN/public_regulated_data_types/archive/master.zip' into '/tmp/pyuavcan-cli-dsdlhybgoa62/dsdl.zip'...
2020-11-03 15:30:47 12985 INFO     pyuavcan._cli.commands.dsdl_generate_packages: Extracting the archive into '/tmp/pyuavcan-cli-dsdlhybgoa62'...
2020-11-03 15:30:47 12985 INFO     pyuavcan._cli.commands.dsdl_generate_packages: Resource 'https://github.com/UAVCAN/public_regulated_data_types/archive/master.zip' contains the following root namespace directories: ['/tmp/pyuavcan-cli-dsdlhybgoa62/public_regulated_data_types-master/uavcan', '/tmp/pyuavcan-cli-dsdlhybgoa62/public_regulated_data_types-master/.vscode', '/tmp/pyuavcan-cli-dsdlhybgoa62/public_regulated_data_types-master/reg']
2020-11-03 15:30:47 12985 INFO     pyuavcan._cli.commands.dsdl_generate_packages: Lookup DSDL root namespace directories: ['/tmp/pyuavcan-cli-dsdlhybgoa62/public_regulated_data_types-master/uavcan', '/tmp/pyuavcan-cli-dsdlhybgoa62/public_regulated_data_types-master/.vscode', '/tmp/pyuavcan-cli-dsdlhybgoa62/public_regulated_data_types-master/reg']
2020-11-03 15:30:47 12985 INFO     pyuavcan._cli.commands.dsdl_generate_packages: Generating DSDL package PosixPath('/home/tomr/foobar/foobar') from root namespace PosixPath('/home/tomr/foobar') with lookup dirs: ['/tmp/pyuavcan-cli-dsdlhybgoa62/public_regulated_data_types-master/uavcan', '/tmp/pyuavcan-cli-dsdlhybgoa62/public_regulated_data_types-master/.vscode', '/home/tomr/foobar', '/tmp/pyuavcan-cli-dsdlhybgoa62/public_regulated_data_types-master/reg']
2020-11-03 15:30:47 12985 INFO     pydsdl._namespace: The namespace at '/home/tomr/foobar' is empty
Error: ValueError: not enough values to unpack (expected 1, got 0)
2020-11-03 15:30:47 12985 INFO     pyuavcan._cli._main: Unhandled exception: not enough values to unpack (expected 1, got 0)
Traceback (most recent call last):
  File "/home/tomr/anaconda3/lib/python3.8/site-packages/pyuavcan/_cli/_main.py", line 24, in main
    exit(_main_impl())
  File "/home/tomr/anaconda3/lib/python3.8/site-packages/pyuavcan/_cli/_main.py", line 50, in _main_impl
    result = args.func(args)
  File "/home/tomr/anaconda3/lib/python3.8/site-packages/pyuavcan/_cli/_main.py", line 146, in execute
    return cmd.execute(args, subsystems)
  File "/home/tomr/anaconda3/lib/python3.8/site-packages/pyuavcan/_cli/commands/dsdl_generate_packages.py", line 111, in execute
    gpi_list = self._generate_dsdl_packages(source_root_namespace_dirs=inputs,
  File "/home/tomr/anaconda3/lib/python3.8/site-packages/pyuavcan/_cli/commands/dsdl_generate_packages.py", line 179, in _generate_dsdl_packages
    gpi = pyuavcan.dsdl.generate_package(root_namespace_directory=ns,
  File "/home/tomr/anaconda3/lib/python3.8/site-packages/pyuavcan/dsdl/_compiler.py", line 188, in generate_package
    root_namespace_name, = set(map(lambda x: x.root_namespace, composite_types))  # type: str,
ValueError: not enough values to unpack (expected 1, got 0)

pavel-kirienko, 2020-11-03:
This is a duplicate of https://github.com/UAVCAN/pyuavcan/issues/130. Would you be inclined to submit a fix, perhaps? Otherwise, I should be able to get to it around the weekend.

closed 2020-11-03 by pavel-kirienko

TomDeRybel, 2020-11-03:
The earliest I can take a look is Thursday (that is, if things go well tomorrow). I'll let you know by Friday if I can take this one on.

TomDeRybel, 2020-11-05:
I took a good look at it, but decided to decline trying to fix this. The main reason is that I'm getting tangled-up in both the tools and (for me) non-trivial code base. It would take me quite a number of days to master all that and I am lacking the time to do that right now.
Thus, if you could be so kind to take a look at this, I would be much obliged.