OpenCyphal/pycyphal #140: Pyuavcan dsdl tools delete source files on error

mawildoer, 2020-12-26:
Hi all, I've been trying to work with Pyuavcan on my raspberry pi, but have run into a persistent issue (or pair). 
1. an error is thrown stating the file I just created and copied for doesn't exist
2. when checking for the file, I've found it's deleted

This is the command in question, but I've run variants of it specifying the output explicitly as well:

```pyuavcan dsdl-generate-packages ./test_dsdl
Error: FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/test_dsdl'```

I've also tested it with the example code from the guide, which produces another error:

```pi@raspberrypi:~ $ pyuavcan dsdl-gen-pkg https://github.com/UAVCAN/public_regulated_data_types/archive/master.zip dsdl_src/my_project
Error: InvalidDirectiveError: /tmp/pyuavcan-cli-dsdlvldd6enq/public_regulated_data_types-master/uavcan/si/unit/frequency/Scalar.1.0.uavcan:2: Unknown directive 'sealed'```

I raised this specific issue regarding the deletion of my source files, which I see as the primary issue.

mawildoer, 2020-12-26:
I have also just tested this on Windows with the latest pull of the git repo. 
Same behaviour as the pi.

pavel-kirienko, 2020-12-28:
Both errors are caused by using an old version of the library. Specifically, the first one is reported in #133 and resolved in #138. The other one is resolved in #128. You should make sure that you are installing the latest version, which is 1.1.0.dev6 at the moment.

closed 2020-12-28 by pavel-kirienko

mawildoer, 2020-12-28:
This seems to be a pair of issues, one being the faulty behaviour and the other being the default installation of a faulty version. 

We've addressed the default installation in my other issue (https://github.com/UAVCAN/pyuavcan/issues/141). 

Thanks!