OpenCyphal/pycyphal #141: Default installation is not functional

mawildoer, 2020-12-26:
First, "--pre" is required to get anything functional for UAVCAN v1 from pip, default is 0.9xx
I would suggest in this case we remove the --pre tag from current releases since no stable really exists. Once it does, it makes sense to reintroduce it.
Alternatively, we can include a clear note of this flag requirement in the instructions, To be effective, I would suggest this note should exist in two places; https://pyuavcan.readthedocs.io/en/latest/pages/installation.html in the "If you need full-featured library, use this and read no more:" code snippet and the README.md page here. 

Secondly, the options for transports and cli aren't as obvious as they could be. I think this will largely be solved once the dsdl functionality transfer to nunavut is finalised, but in the meantime, perhaps it's also worth a note in the README?

I figure we can have a discussion here and I'm happy to implement the changes I can once they're agreed.

pavel-kirienko, 2020-12-28:
1. The stable v1.1 will be released in the following 2 days after https://github.com/UAVCAN/pyuavcan/pull/139 is merged.

2. The option for CLI will be removed very soon because the CLI tool will no longer be part of PyUAVCAN (see #134). Could you please elaborate on why the transport-specific options are non-obvious?

mawildoer, 2020-12-28:
1. Given that kinda timeframe, no worries. I guess this is nice to have for posterity sake and anyone else who may have the same headache and comes across it. 

2. I did also notice the CLI was going to be extracted but figured that the overhead to make a note of this option would be low. In my experience when working with python packages, the default installation is relatively fully featured. It's a bit unintuitive to need additional tags for an installation for only "basic" usage, especially since these tags are only referenced in one location.

commit 8bc9c4b9 (2020-12-31): Release the first stable version of the library.

This is v1.1, not v1.0, due to historical reasons.
Fixes #141
closed 2020-12-31 by pavel-kirienko