OpenCyphal/pycyphal #142: Numpy ~=1.19 requirement is unnecessarily restrictive

alkasm, 2021-01-03:
The numpy version requirement was bumped from ~=1.16 to ~=1.19 to support the `bitorder` kwarg in `np.packbits()` in https://github.com/UAVCAN/pyuavcan/pull/124/commits/e24d2ce3665b03eb4638c97c3965a48bd98b4e91, although this kwarg was introduced in numpy 1.17 (the [release notes](https://numpy.org/doc/stable/release/1.17.0-notes.html#packbits-and-unpackbits-accept-an-order-keyword) incorrectly calls the kwarg `order` but the function [docs](https://numpy.org/doc/1.17/reference/generated/numpy.packbits.html) specify it correctly).

Not really a big issue but if there isn't a specific need for 1.19, there's no reason for this library to require it.

pavel-kirienko, 2021-01-03:
Does this requirement create any specific issues in your case? If not, I am inclined to keep it as-is to reduce the variability of production environments.

alkasm, 2021-01-03:
Not any unworkable one, I can patch the requirement for example in my case. OTOH I felt it kept with the ethos of minimal dependencies to also require the minimum viable version. Since numpy changed some of its internals to be able to use 64-bit BLAS and LAPACK libs in 1.18 and removed support for Python < 3.6 in 1.19, it could be limiting for certain toolchains. OTOH I couldn't find any documentation for PyUAVCAN specifying the minimal Python version targeted, so Python 3.5 and lower support might not be interesting to you.

pavel-kirienko, 2021-01-03:
The minimal required version is not explicitly documented anywhere but we enforce this:

https://github.com/UAVCAN/pyuavcan/blob/8bc9c4b936b36642a6da3723c4013eb367c29464/pyuavcan/__init__.py#L62-L63

If you prefer to lower the minimal version requirement down to 1.18 or 1.17, a pull request would be accepted.

closed 2021-01-03 by pavel-kirienko