OpenCyphal/pycyphal #178: feature: Please Add canalystii to pyuavcan/transport/can/media/pythoncan

Cherish-forever, 2021-08-30:
python-can support canalystii device, I think pyuavcan need support it

pavel-kirienko, 2021-08-30:
Indeed, a pull request would be welcome.

Cherish-forever, 2021-08-30:
@pavel-kirienko 
I have tried, but failed.
I have add
`def _construct_canalystii(parameters: _InterfaceParameters) -> can.ThreadSafeBus:
    if isinstance(parameters, _ClassicInterfaceParameters):
        return can.ThreadSafeBus(interface=parameters.interface_name, bitrate=parameters.bitrate)
    if isinstance(parameters, _FDInterfaceParameters):
        return can.ThreadSafeBus(
            interface=parameters.interface_name,
            channel=parameters.channel_name,
            bitrate=parameters.bitrate,
        )
    assert False, "Internal error"`
and
`        "canalystii": _construct_canalystii,`
to pyuavcan/transport/can/media/pythoncan/_pythoncan.py

then
`export YAKUT_TRANSPORT="CAN(can.media.pythoncan.PythonCANMedia(\"canalystii:0\", 1_000_000), 42)"`

but failed:
`$ Error: Invalid value: Could not initialize transport 'CAN(can.media.pythoncan.PythonCANMedia("canalystii:0", 1_000_000), 42)': TypeError("__init__() missing 1 required positional argument: 'channel'")`

pavel-kirienko, 2021-08-30:
@Polarisru do you happen to have a suggestion here?

Cherish-forever, 2021-08-30:
@pavel-kirienko @Polarisru 
I am using yakut

pavel-kirienko, 2021-08-30:
@Cherish-forever after you reformatted the source properly, it is obvious that you forgot to pass `channel=parameters.channel_name,` in the Classic case. Fix that and see if it works.

Usage of `YAKUT_TRANSPORT` is generally discouraged, consider using `UAVCAN__CAN__IFACE` instead as suggested in the README. This is not related to the current problem though, just a general suggestion.

Cherish-forever, 2021-08-30:
@pavel-kirienko 
I have been run it pass.

here is my change:
```
def _construct_canalystii(parameters: _InterfaceParameters) -> can.ThreadSafeBus:
    if isinstance(parameters, _ClassicInterfaceParameters):
        return can.ThreadSafeBus(interface=parameters.interface_name, bitrate=parameters.bitrate)
    if isinstance(parameters, _FDInterfaceParameters):
        return can.ThreadSafeBus(
            interface=parameters.interface_name,
            channel=parameters.channel_name,
            bitrate=parameters.bitrate[0],
        )
    assert False, "Internal error"
```

and download canalystii so lib:
https://github.com/AbangLZU/can_control/blob/master/can_control_node/lib/libcontrolcan.so

because:
https://github.com/hardbyte/python-can/blob/develop/can/interfaces/canalystii.py#L65

so lib need put in current run path.

then you can use canalystii device.

oh here is my enverioment:
```
export YAKUT_COMPILE_OUTPUT=~/.yakut
export YAKUT_PATH="$YAKUT_COMPILE_OUTPUT"
export UAVCAN__NODE__ID=42
export UAVCAN__CAN__IFACE="canalystii:0"
export UAVCAN__CAN__MTU="8"
export UAVCAN__CAN__BITRATE=1000000
```

pavel-kirienko, 2021-08-30:
The FD case discards the data segment bit rate, which is incorrect:

> `bitrate=parameters.bitrate[0]`

Cherish-forever, 2021-08-30:
@pavel-kirienko 
CANalyst-II MCU is PIC32MX795F512L 2 x CAN2.0b modules with 1024 buffers
here is the detail:
https://www.microchip.com/en-us/product/PIC32MX795F512L

not support can fd.

pavel-kirienko, 2021-08-30:
Then you should throw an error from the FD case instead of defaulting to the Classic mode.

Cherish-forever, 2021-08-30:
@pavel-kirienko 
like this?
```
def _construct_canalystii(parameters: _InterfaceParameters) -> can.ThreadSafeBus:
    if isinstance(parameters, _ClassicInterfaceParameters):
        return can.ThreadSafeBus(interface=parameters.interface_name, bitrate=parameters.bitrate)
    if isinstance(parameters, _FDInterfaceParameters):
        return can.ThreadSafeBus(
            interface=parameters.interface_name,
            channel=parameters.channel_name,
            bitrate=parameters.bitrate[0],
        )
    if isinstance(parameters, _FDInterfaceParameters):
        raise InvalidMediaConfigurationError(f"Interface does not support CAN FD: {parameters.interface_name}")
    assert False, "Internal error"
```

pavel-kirienko, 2021-08-30:
The second instance check is incorrect, otherwise yes.

Cherish-forever, 2021-08-30:
@pavel-kirienko 
I have some confuse:
why I have do:
```
export YAKUT_COMPILE_OUTPUT=~/.yakut
export YAKUT_PATH="$YAKUT_COMPILE_OUTPUT"
export UAVCAN__NODE__ID=42
export UAVCAN__CAN__IFACE="canalystii:0"
export UAVCAN__CAN__MTU="8"
export UAVCAN__CAN__BITRATE=1000000
```
I got parameters is _FDInterfaceParameters, something hidden in my eye.

pavel-kirienko, 2021-08-30:
Can you rephrase the question, please?

pavel-kirienko, 2021-08-31:
>pass bitrate is (br_arb, br_data), this meas that device support CAN FD ?

It doesn't say anything about the capabilities of the device itself. It only means that if arbitration != data bitrate, the caller is expecting the device to support CAN FD.

Cherish-forever, 2021-09-01:
@pavel-kirienko 
Thank you very much, I got it.
```
export UAVCAN__CAN__BITRATE="1000000 1000000"
```

and 
```
def _construct_canalystii(parameters: _InterfaceParameters) -> can.ThreadSafeBus:
    if isinstance(parameters, _ClassicInterfaceParameters):
        return can.ThreadSafeBus(interface=parameters.interface_name,
                                 channel=parameters.channel_name,
                                 bitrate=parameters.bitrate)
    if isinstance(parameters, _FDInterfaceParameters):
        raise InvalidMediaConfigurationError(f"Interface does not support CAN FD: {parameters.interface_name}")
    assert False, "Internal error"
```

Cherish-Gww, 2021-09-01:
@pavel-kirienko 
Hi, I have PR for this feature, and this is my first commit to open source project, and I don`t know how to add reviewers.

closed 2021-09-01 by pavel-kirienko

Cherish-Gww, 2021-09-03:
pyuavcan base on python-can If you want use CANalyst-II, you need install python-can package.
I have contact to CANalyst-II manufacturer, he does not provide source code to python-can, so you need download CANalyst library or you can install python-can by this:
```
$ pip3 install git+https://github.com/Cherish-Gww/python-can.git@add_canalystii_so
```
it will automatic install libraries to your local.

pavel-kirienko, 2021-09-03:
@Cherish-forever Consider sending a pull request adding these notes to the docs, maybe? This place is not discoverable at all for users, so very few people are going to see what you just wrote.

Cherish-Gww, 2021-09-03:
> @Cherish-forever Consider sending a pull request adding these notes to the docs, maybe? This place is not discoverable at all for users, so very few people are going to see what you just wrote.

like this?

```
            - Interface ``canalystii`` is described in
              https://python-can.readthedocs.io/en/stable/interfaces/canalystii.html.
              You need download CANalyst library for python-can package or you can install python-can by:
              ``pip3 install git+https://github.com/Cherish-Gww/python-can.git@add_canalystii_so``
              Example: ``canalystii:0``
```

pavel-kirienko, 2021-09-03:
Yep, this seems sensible. Are you planning to have your fork merged into upstream of Python-CAN?

Cherish-Gww, 2021-09-03:
> Yep, this seems sensible. Are you planning to have your fork merged into upstream of Python-CAN?

https://github.com/hardbyte/python-can/pull/1123
they don`t want add the binary DLLs/SOs to python-can.

MKS-Howard, 2021-09-08:
The DLL in your library is 64-bit. If you use python-32bit, ControlCAN.dll must be changed to 32-bit or python-64bit.

Cherish-Gww, 2021-11-09:
Good news：

This driver is based on black box reverse engineering of the USB behaviour of the proprietary software, and reading the basic data structure layouts in the original python-can canalystii source.

https://github.com/hardbyte/python-can/commit/18f346211bf937593fbe2304d3f850be2474a129
hosted at
https://github.com/projectgus/python-canalystii

python-can 4.0.0 will solve this issue. but not now.

commit bd4123e4 (2021-09-01): UAVCAN/CAN: Extend the PythonCAN media layer with CANalyst-II device support (#179)

Fixes #178