OpenCyphal/pycyphal #182: [UAVCAN v0] GitHub Actions CD workflow should only be run on `legacy-v0`, not on feature branches

pavel-kirienko, 2021-09-07:
The workflow just published a release from my feature branch at https://github.com/UAVCAN/pyuavcan/pull/181 :scream_cat:

coderkalyan, 2021-09-08:
Considering https://github.com/UAVCAN/pyuavcan/issues/172 was closed, do you still want me to address this, or just disable GH actions? Either works for me, its not hard to specify the branches,

pavel-kirienko, 2021-09-08:
I understood that #172 was intended for v1, which is impractical to work on right now, so I closed it. For v0 we are already using Actions so let's keep that. Please do specify the correct branches.

coderkalyan, 2021-09-11:
Fixed by https://github.com/UAVCAN/pyuavcan/commit/acf1e158a5224fb969117b9c96119ab767886090

closed 2021-09-11 by coderkalyan