OpenCyphal/pycyphal #229: UDP subscriptions are broken on macOS

pavel-kirienko, 2022-05-12:
Reported by Lydia at https://matrix.to/#/!iIoEluFjwtvXWxhOJg:matrix.org/$9yephH_8t7U6wxhbOtes4dSS6AOTQAugWQE1NtENcoY:

```
lydiagh@147dda273336 ~ % export UAVCAN__UDP__IFACE=127.11.0.0
lydiagh@147dda273336 ~ % yakut monitor
InvalidMediaConfigurationError: Could not register multicast group membership 239.11.29.85 via 127.11.0.0 using <socket.socket [closed] fd=-1, family=[AddressFamily.AF](http://addressfamily.af/)_INET, type=SocketKind.SOCK_DGRAM, proto=17> [EADDRNOTAVAIL]
```

The relevant code is here:

https://github.com/OpenCyphal/pycyphal/blob/1c4539390016eafb87b8377259b425883f19ee2e/pycyphal/transport/udp/_ip/_v4.py#L104-L115

Multicast handling differs rather significantly between GNU/Linux and Windows. It is possible that yet another, macOS-specific case, needs to be introduced.

emrainey, 2022-05-12:
In my environment (Mac) it fails on line 115 above during `IP_ADD_MEMBERSHIP`. It does not throw during `bind`.
I added some code surrounding this to also set `SO_REUSEPORT` and to bind w/ the multicast IP address on mac.

```python
        s.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)
        if sys.platform.startswith("linux") or sys.platform.startswith("darwin"):  # pragma: no branch
            # This is expected to be useful for unicast inputs only.
            # https://stackoverflow.com/a/14388707/1007777
            s.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEPORT, 1)

        if isinstance(data_specifier, MessageDataSpecifier):
            multicast_ip = message_data_specifier_to_multicast_group(self._local, data_specifier)
            multicast_port = SUBJECT_PORT
            if sys.platform.startswith("linux") or sys.platform.startswith("darwin"):
                # Binding to the multicast group address is necessary on GNU/Linux: https://habr.com/ru/post/141021/
                print('Binding w/ Multicast')
                s.bind((str(multicast_ip), multicast_port))
            else:
                # Binding to a multicast address is not allowed on Windows, and it is not necessary there. Error is:
                #   OSError: [WinError 10049] The requested address is not valid in its context
                print('Binding w/o Multicast')
                s.bind(("", multicast_port))
            try:
                # Note that using INADDR_ANY in IP_ADD_MEMBERSHIP doesn't actually mean "any",
                # it means "choose one automatically"; see https://tldp.org/HOWTO/Multicast-HOWTO-6.html
                # This is why we have to specify the interface explicitly here.
                packed_data = multicast_ip.packed + self._local.packed
                print(f'Using {multicast_ip} and {self._local} ==> {packed_data.hex()}')
                s.setsockopt(socket.IPPROTO_IP, socket.IP_ADD_MEMBERSHIP, multicast_ip.packed + self._local.packed)
            except OSError as ex:
                s.close()
                if ex.errno in (errno.EADDRNOTAVAIL, errno.ENODEV):
                    raise InvalidMediaConfigurationError(
                        f"Could not register multicast group membership {multicast_ip} via {self._local} using {s} "
                        f"[{errno.errorcode[ex.errno]}]"
                    ) from None
                raise  # pragma: no cover
```

pavel-kirienko, 2022-06-20:
Fixed in https://github.com/OpenCyphal/pycyphal/pull/230