OpenCyphal/pycyphal #248: Fixing demo issues (MacOS)

maksimdrachov, 2022-09-25:
I'm trying to run the PyCyphal demo (as discussed [here](https://pycyphal.readthedocs.io/en/stable/pages/demo.html)).

I'm doing the following:

```
mkdir ~/pycyphal-demo
cd ~/pycyphal-demo
python3 -m venv .venv
source .venv/bin/activate
```

### Issue 1: Installing `pycyphal`

[image]

So instead I did: (which seems to work)

```
pip3 install pycyphal
```

### Issue 2: DSDL definitions path

The 2 files are created in path `~/pycyphal-demo/sirius_cyber_corp` initially.

Then in the next section, this changes slightly:

```
custom_data_types/
    sirius_cyber_corp/                          # Created in the previous section
        PerformLinearLeastSquaresFit.1.0.dsdl
        PointXY.1.0.dsdl
public_regulated_data_types/                    # Clone from git
    uavcan/                                     # The standard DSDL namespace
        ...
    ...
demo_app.py                                     # The thermostat node script
```

I assume this `custom_data_types` folder is unnecessary?

### Issue 3: No module named `uavcan`

Now, when I try to run `demo_app.py`, I'm getting the following error:

[image]

Btw, I have the same issue in Ubuntu:

[image]

pavel-kirienko, 2022-09-25:
1. zsh attempts to interpret your `[]` as a glob pattern rather than a literal command, so you should use quotes:

```
pip install 'pycyphal[transport-can-pythoncan,transport-serial,transport-udp]'
```

2. It is recommended to keep DSDL namespace directories inside a dedicated directory, such as `custom_data_types` in this case, to ensure that the DSDL compiler does not attempt to compile directories that are not DSDL namespaces. It may not make much difference in this specific case but in general it is a good practice to follow.

3. I think it might be a regression in the demo caused by a recent change introduced in https://github.com/OpenCyphal/pycyphal/pull/236. Could you please check if exporting this fixes the problem:

```
export CYPHAL_PATH="$HOME/pycyphal-demo/custom_data_types:$HOME/pycyphal-demo/public_regulated_data_types"
```

This environment variable is supposed to contain a list of paths where the DSDL root namespace directories are to be found. When your Python interpreter encounters an import statement referring to a non-existent module, PyCyphal will go check if any of the paths listed in `CYPHAL_PATH` contain a DSDL namespace named like the missing module. If one is found, it will (re)compile all available DSDL namespaces and import the resulting Python package.

This mechanism used to work differently until the recent PR I linked above; we updated the docs accordingly but apparently the demo still requires some work. If this solution fixes the problem, a pull request amending the demo would be very welcome.

maksimdrachov, 2022-10-01:
Ok, first 2 issues are understood.

On the third issue, I'm still getting the following error:

[image]

My understanding of this error:

`make_node` calls `make_transport`, which should "construct a transport instance based on the configuration encoded in the supplied registers".

The registry in this case is `demo_app.db`:

[image]

For some reason `make_transport` is returning `None`.

[image]

maksimdrachov, 2022-10-01:
So I have my `demo_app.py` running:

[image]

But then trying to connect with `yakut`, I'm getting the following error:

[image]

maksimdrachov, 2022-10-01:
If I run with `sudo`, the error changes:

[image]

maksimdrachov, 2022-10-01:
With Yakut Orchestrator:

[image]

pavel-kirienko, 2022-10-01:
Please set the node-ID by exporting this:

```bash
export UAVCAN__NODE__ID=123  # Pick an arbitrary value in [0,4094]
```

In the future you can avoid the hassle of exporting env vars manually by simply sourcing something like this:

```bash
export UAVCAN__UDP__IFACE='127.9.0.0'
export UAVCAN__NODE__ID=$(yakut accommodate)
echo "Auto-selected node-ID for this session: $UAVCAN__NODE__ID"
```

pavel-kirienko, 2022-10-01:
> If I run with sudo, the error changes

This is because sudo runs the command as a different user from a new session that does not inherit environment variables from the current session:

```bash
~ ❯ export ABC=123
~ ❯ bash -c 'echo $ABC'
123
~ ❯ sudo bash -c 'echo $ABC'

```

pavel-kirienko, 2022-10-01:
Can you please confirm that this suggestion worked for you:

> 3. I think it might be a regression in the demo caused by a recent change introduced in https://github.com/OpenCyphal/pycyphal/pull/236. Could you please check if exporting this fixes the problem: <...>

If yes, would you mind helping us fix the guide?

closed 2022-10-09 by maksimdrachov