OpenCyphal/pycyphal #251: Bug in redundant transport

silverv, 2022-10-06:
```
Traceback (most recent call last):
  File "C:\Users\silver\zubax\yukon\venv\lib\site-packages\pycyphal\application\heartbeat_publisher.py", line 203, in _task_function
    if not await pub.publish(self.make_message()):
  File "C:\Users\silver\zubax\yukon\venv\lib\site-packages\pycyphal\presentation\_port\_publisher.py", line 113, in publish
    return await self._maybe_impl.publish(message, self._priority, loop.time() + self._send_timeout)
  File "C:\Users\silver\zubax\yukon\venv\lib\site-packages\pycyphal\presentation\_port\_publisher.py", line 197, in publish
    return await self.transport_session.send(transfer, monotonic_deadline)
  File "C:\Users\silver\zubax\yukon\venv\lib\site-packages\pycyphal\transport\redundant\_session\_output.py", line 273, in send
    assert not results or all(isinstance(x, bool) for x in results)
AssertionError
```
I think it is related to the raising of this exception in my backend tests for Yukon
```
class FaultyOutputSession(OutputSession, FaultySession):
    def __init__(self, inner: OutputSession):
        self._inner = inner
        self._closed = False

    @property
    def specifier(self) -> pycyphal.transport.InputSessionSpecifier:
        return self._inner._specifier

    def sample_statistics(self) -> TransportStatistics:
        raise NotImplementedError

    async def send(self, transfer: Transfer, monotonic_deadline: float) -> bool:
        if self.should_fail():
            raise pycyphal.transport.ResourceClosedError("The transport is closed")
        return self._inner.send(transfer, monotonic_deadline)
```

pavel-kirienko, 2022-10-06:
If you manage to reproduce it again, could you please share the values of the involved objects here:

https://github.com/OpenCyphal/pycyphal/blob/6ebf1167a37292f3f6bcd9fdb4fc428344177442/pycyphal/transport/redundant/_session/_output.py#L273

silverv, 2022-10-06:
I did not await my sends

closed 2022-10-15 by pavel-kirienko