OpenCyphal/pycyphal #277: Migrate to Nunavut v2

pavel-kirienko, 2023-03-07:


Willmac16, 2023-12-18:
Is there a list of remaining tasks to move to Nunavut v2?

I would be happy to help if I can, as my build system would be greatly simplified without two different versions of Nunavut to wrangle.

pavel-kirienko, 2023-12-23:
List of tasks to move to Nunavut v2:

Remove the DSDL templates completely and instead invoke the default Python generator available in Nunavut v2. These directories need to go: `pycyphal/dsdl/_templates`, `pycyphal/dsdl/_serialized_representation`. `_builtin_form.py`, `_composite_object.py` are not needed either because the respective functionality has been moved into the Nunavut-generated support module.

Upgrade `_compiler.py` to invoke the new Nunavut API correctly.

Remove all DSDL-related tests affected by the removal of the modules mentioned above. They have already been migrated to Nunavut.

commit d02f84e3 (2024-01-08): Migrate to Nunavut v2 (#318)

Closes #277
closed 2024-01-08 by pavel-kirienko