OpenCyphal/pycyphal #286: Is very slow on Windows

silverv, 2023-03-24:
Perhaps causes #281 and also affects Yukon usability on Windows devices.

pavel-kirienko, 2023-03-29:
@silverv we should try increasing the resolution of the system timer as done in the PyCyphal test suite:

https://github.com/OpenCyphal/pycyphal/blob/029fdb5b5a158372103edf436298bc3559377dd8/tests/conftest.py#L63

More on this in the Windows API docs

silverv, 2023-03-31:
It would probably make sense to go for a lower than 6 ms value, ideally a lower than millisecond value

silverv, 2023-03-31:
I don't understand why setting the delay even even higher helped the tests

pavel-kirienko, 2023-03-31:
Not sure what delay you are referring to. There are no delays here, the referenced API function changes the system timer resolution. It should be possible to go down to 1 ms, I would try that.

closed 2024-01-05 by pavel-kirienko