OpenCyphal/pycyphal #300: Do we need NCAT.exe ?

emrainey, 2023-06-01:
This was flagged by a corporate security audit when I downloaded pycyphal. While I'm on a Mac and I can't directly run this, it does beg the question, why is this needed here?

pavel-kirienko, 2023-06-02:
It is necessary for testing Cyphal/serial on Windows. On GNU/Linux, it is installed as a package during CI workflow execution, but there is no clean way of doing so for Windows that I could find. Here is how it's used:

https://github.com/OpenCyphal/pycyphal/blob/de11875b1ec66eb22e184d8edb0249cd70ae0f72/noxfile.py#L18-L20

https://github.com/OpenCyphal/pycyphal/blob/de11875b1ec66eb22e184d8edb0249cd70ae0f72/noxfile.py#L78

If you have a better solution in mind, please send a pull request; for now I am closing this. Also, keep in mind that the default Windows security policies tend to misidentify `ncat.exe` as a "hacking tool"...

thirtytwobits, 2023-06-02:
There's got to be a python solution available here?

thirtytwobits, 2023-06-02:
@emrainey , do you think we could replace this function with a Python library?

emrainey, 2023-06-15:
Likely we could with a raw socket over UDP? Looks like there are some minor implementation on stack overflow. https://stackoverflow.com/questions/1908878/netcat-implementation-in-python. 

ChatGPT could probably pound out an implementation in no time. :)

pavel-kirienko, 2023-06-16:
This is just a TCP broker that bridges all connections accepted through some specified port; it can be implemented easily. Note that it has to be TCP, not UDP.

pavel-kirienko, 2023-06-16:
While we're at it, we should also use zero-byte frame delimiters on each connection to prevent output-queue mangling. This is an issue with ncat.exe that complicates the testing of larger networks.

closed 2025-07-11 by pavel-kirienko