OpenCyphal/pycyphal #309: Environment Variables Suck

thirtytwobits, 2023-08-02:
Many many reasons why but here are two things that need to be fixed:

1. Always recheck and rebuild the database when environment variables change. For example, I just wasted 15 minutes trying to figure out why pycyphal was trying to connect to UDP only to discover that my previous value UAVCAN__UDP__IFACE was somehow part of the db. Once I deleted this and restarted my new environment variables were found.
2. Always accept "CYPHAL_" where "UAVCAN_" is currently required. It's really hard to remember which of the environment variables are UAVCAN and which are CYPHAL. Either should be accepted but CYPHAL should be documented.

pavel-kirienko, 2023-08-02:
1. We can make it so that environment variables override all entries in the database during initialization, like the following but executed for all records in the db during construction:

  https://github.com/OpenCyphal/pycyphal/blob/01b9a9b57143bc916bfa2fa26102a787b4e558e9/pycyphal/application/register/_registry.py#L256-L261

2. This is outside of the scope of PyCyphal because the names come from https://github.com/OpenCyphal/public_regulated_data_types/blob/935973babe11755d8070e67452b3508b4b6833e2/uavcan/register/384.Access.1.0.dsdl#L80

pavel-kirienko, 2025-05-12:
Superseded by https://github.com/OpenCyphal/pycyphal/issues/351

closed 2025-05-12 by pavel-kirienko