OpenCyphal/pycyphal #338: Add comment explaining how to properly import types

maksimdrachov, 2024-05-05:
`sirius_cyber_corp` example shows how to work with types in the top directory, however if the type is located in a folder, one has to make sure to import the type as follows:

```py
import top_folder.sub_folder.some_type # import like this
top_folder.sub_folder.some_type # use like this

# This won't work
import top_folder
top_folder.sub_folder.some_type # This will give an error
```

[image]

pioneerAlone, 2025-02-17:
I got also the same issue, I cannot import **sirius_cyber_corp** and  **uavcan.node**  as shown in the demo even if I set the EVN variable for CYPHAL_PATH.

How can I resolve it?

maksimdrachov, 2025-02-22:
Please use the forum for questions: https://forum.opencyphal.org/

Don't forget to also provide some additional information on what you've tried so far.

pioneerAlone, 2025-02-26:
Thanks ,I've resolved this.

Regards,
Bakewell