OpenCyphal/pycyphal #359: Warning issued for multiple frame delimiters/zero bytes in stream for serial transport

hannesweisbach, 2025-06-25:
Hi!

I'm using pycyphal 1.23.0 with serial transport. I know the serial transport specification is still marked "experimental".

I accidentally sent consecutive zero bytes (frame delimiters) during development, which should be allowed according to the Cyphal Specification, section 4.2.2. "The number of frame delimiters between adjacent frames may exceed one.".

However, the pycyphal serial transport issues a warning about this:
```
2025-06-25 12:39:19,552 - pycyphal.transport.serial._serial - WARNING - socket://127.0.0.1:4444: Out-of-band received at 2025-06-25T12:39:19.548269/7172.518125: '\x00\x00'
```

I think for a behaviour that is explicitly allowed as per the specification, no warning should be issued.

pavel-kirienko, 2025-06-25:
Ah yes the warning should be demoted to INFO or DEBUG. The library is trying to say that it suspects that there might be some other protocol using the same serial link.

pavel-kirienko, 2025-06-25:
A pull request demoting the warning would be accepted.

pavel-kirienko, 2025-07-01:
Done in #361